freeacs/freeacs #307: Change url services

fernandezteran, 2019-03-20:
Good afternoon,
I have 20,000 devices with ACS URL: http://cagada.es/web/index

I tried
[Change url services #302](https://github.com/freeacs/freeacs/issues/302)

,but it does not work. Can somebody help me? thanks

pingwiniasty, 2019-03-20:
If I understand, you want to change ACS URL on your devices?

fernandezteran, 2019-03-20:
Goodnight. no, I want to configure my server tr069 with that url (http://cagada.es/web/index). I don’t want to use the url that it has by default (http://localhost:8085/tr069) That's possible?

pingwiniasty, 2019-03-21:
If want to access to the web module by the same url, than no.

jarlah, 2019-03-21:
I guess what you want is to use a publicly accessible domain for the ACS url? If so, you need nginx or apache or haproxy (or similar) that can proxy the requests into your server. We gave you the nginx config in the previous issue. In addition you must/should generate a ssl certificate to protect communication, if its public. See wiki for that.

There is no reason to change the ACS url from /tr069 to /web/index. It makes no sense and only confuses us if we are to help. /web is for admin gui where you administer your devices. /tr069 is an edpoint where units post soap messages (xml). 

I suggest. If you have installed freeacs on a server. Make sure you have installed nginx on the same server. By default it listens on port 80 and if you have configured ssl also on port 443 (https). It should be installed if using script. You could potentially use different context path for each module but i dont see why.

fernandezteran, 2019-03-21:
> I guess what you want is to use a publicly accessible domain for the ACS url? If so, you need nginx or apache or haproxy (or similar) that can proxy the requests into your server. We gave you the [...]

Hi Jarlah. 
the reason i must change that is that i have 20k, more or less, cpe´s that point to a wrong url... our provider does a great mistake with that and puted a wrong url in these cpe´s making them go to http://myACSserver/web/index instead http://myACSserver/tr06

jarlah, 2019-03-21:
Ah ok. you need a proxy in front that can reroute. So it did not work to lets say modify the nginx config? Can you show us the config you have here? i know from Experience that hacking on nginx config can be pretty difficult.

jarlah, 2019-03-21:
We will fix this :)

fernandezteran, 2019-03-22:
Hi jarlah,
Attachment configuration:
[image]

Thanks!

jarlah, 2019-03-23:
The nginx config above is not doing what you want. Ill show you why in a moment.

jarlah, 2019-03-23:
In your config you are saying that inbound requests to /tr069 should be proxied to the path /web/index on the tr069 server. There is no path /web/index on the tr069 server so this will never work. 

You need to change the location paths, and not the proxy pass inside the location config, so that requests to location /web/index is proxied to the tr069 server. I also suggest duplicating the tr069 location, and then provision each of the 20K devices to change their provision urls. And finally remove the hack.

```
events {
  worker_connections  19000;
}

http {
  server {
    listen       80;
    server_name  localhost;
    proxy_set_header Host $host:$server_port;

    # For the geeks: "A man is not dead while his name is still spoken." -Terry Pratchett
    add_header X-Clacks-Overhead "GNU Terry Pratchett";
    location = / {
      return 301 /web/index;
    }
    location /tr069 {
      proxy_set_header        X-Real-IP       $remote_addr;
      proxy_pass http://localhost:8085/tr069;
    }
    location /web/index {
      proxy_set_header        X-Real-IP       $remote_addr;
      proxy_pass http://localhost:8085/tr069;
    }
    location /web/gui/ {
      proxy_pass http://localhost:8081/web/;
    }
    location /monitor/ {
      proxy_pass http://localhost:8090/monitor/;
    }
    location /webservice/ {
      proxy_pass http://localhost:8088/webservice/;
    }
    location /syslog/ {
      proxy_pass http://localhost:8086/syslog/;
    }
    location /core/ {
      proxy_pass http://localhost:8083/core/;
    }
    location /stun/ {
      proxy_pass http://localhost:8087/stun/;
    }
  }
}
```
as you can see I have renamed /web to /web/gui, but I cannot guarantee that this will work. A better fix would be to update the application-config.conf in /opt/freeacs-web/config with a different context path, for ex admin, and do this:

```
    location /admin/ {
      proxy_pass http://localhost:8081/admin/;
    }
```

However I am not completely sure that everything relies on relative paths, so it might try to redirect to /web somewhere sometime. 

Try it out

jarlah, 2019-03-23:
Remember to restart the nginx service with systemctl restart nginx or services nginx restart (or whatever you use for service management)

cuongdoannguyen, 2020-07-31:
Hi Jarlah,
Thank you for great freeacs and this solution.
I implemented the solution you provide and it works for Genexis and huawei CPE's.
but AVM FritzBox CPE's doesn't accept redirect 301 code. When they hit redirect 301 code they disconnect the session and establish a new session again.

In my case all the cpe has pre-configure acs url://  http://domainname/ 
It is possible to change root location to tr069/ location?
I am trying to resolve this with a haproxy, but can get it working yet. Any suggestion is wellcome.!
Regards,
Doan

closed 2021-09-04 by jarlah