freeacs/freeacs #309: reboot CPE

zzzacbbt, 2019-04-09:
I want to reboot CPE by freeacs server 
I add this job 
[image]
but It was not reboot immediately until CPE send request.
How to control CPE reboot immediately use by tr069_script.

jarlah, 2019-04-10:
I think you would need to perform a kick. I dont remember how or if its possible in script. I will maybe check it out. But my gut feeling is that rebooting a fleet of cpes is not possible without configuring advanced properties to be able to reach those cpes via tcp or udp, in a profile or in each unit.

zzzacbbt, 2019-04-11:
Hi .jarlah.I don't know how to perform a kick.Could you tell me?

pingwiniasty, 2019-04-11:
@zzzacbbt I will check it tomorrow.

zzzacbbt, 2019-04-12:
OK,thank you @pingwiniasty

pingwiniasty, 2019-04-12:
@zzzacbbt Please attach here tr069-conversation.log and service windows configuratio. You can check it in Unit profile -> Select Action -> Service Window.

zzzacbbt, 2019-04-12:
[image]

zzzacbbt, 2019-04-12:
How to get tr069-conversation.log

techdpnn, 2019-04-12:
default.
`/opt/freeacs-tr069/logs`

zzzacbbt, 2019-04-12:
[tr069-conversation.log](https://github.com/freeacs/freeacs/files/3072285/tr069-conversation.log)

zzzacbbt, 2019-04-12:
[devicestatus.zip](https://github.com/freeacs/freeacs/files/3072324/devicestatus.zip)
After CPE INFROM,It did the script and reboot,and always do something like this.
should I re-install the server?

techdpnn, 2019-04-12:
i want to discuss with you  about freeacs ,could you please leave a email or something else  to contact with you .thanks

techdpnn, 2019-04-12:
@zzzacbbt  i want to ask you some question about freeacs ,could you please leave a email or something else  to contact with you .thanks

zzzacbbt, 2019-04-12:
13122231591@163.com

dariosky75, 2019-04-30:
@zzzacbbt
Hi, are you able to kick a single device from the web console? I think you should be first getting this done before trying the other

zzzacbbt, 2019-04-30:
I don't understand how to kick a device,Do you mean delete the unit?

dariosky75, 2019-05-02:
@zzzacbbt 
I am a newbie on the subject so bear with me. 
KICK basically means that your are pinging the CPE to tell "get back in touch with the ACS and provision" otherwise you will have to wait periodic inform Interval set on the CPE.
For KICK to work  parameters should be properlu configured(authentication).
If you go to a specific unit configuration something like   
XXXXXXX/web/index?page=unit-configuration&init_refreshpage=Refresh
You see the provisioning sections all of the 4 commands in there perform a KICK to begin with
Provision, ReadAll, Reboot, Reset

[image]

jarlah, 2019-05-02:
To get this working satisfactory, you should set connectionrequest[url|username|password] unittype parameters to RA. Some cpes doesnt reply with other than blank on these. If the cpe doesnt send the values of these parameters, i often feel its very difficult to get around the problem and i am working to find a solution for this. In the mean time its possible to set a series of system parameters to enable the acs to build up a dynamic kick url. But its a bit flaky at the moment.

closed 2021-09-04 by jarlah