freeacs/freeacs #310: Monitor service accumulates CLOSE_WAIT sockets under HTTP to HTTPS redirect

Daniel-Beardsmore, 2019-04-11:
I have noticed that the freeacs-monitor accumulates CLOSE_WAIT sockets:

 root@acs:~# lsof | grep "CLOSE_WAIT" | head -n 10
 java      21591               freeacs   68u     IPv6           12993544       0t0        TCP localhost:58225->localhost:https (CLOSE_WAIT)
 java      21591               freeacs   69u     IPv6           13008854       0t0        TCP localhost:52741->localhost:https (CLOSE_WAIT)
 java      21591               freeacs   70u     IPv6           13056413       0t0        TCP localhost:52165->localhost:https (CLOSE_WAIT)
 java      21591               freeacs   71u     IPv6           12995503       0t0        TCP localhost:42965->localhost:https (CLOSE_WAIT)
 java      21591               freeacs   72u     IPv6           12999516       0t0        TCP localhost:43357->localhost:https (CLOSE_WAIT)
 java      21591               freeacs   73u     IPv6           13024874       0t0        TCP localhost:48867->localhost:https (CLOSE_WAIT)
 java      21591               freeacs   75u     IPv6           13053442       0t0        TCP localhost:44073->localhost:https (CLOSE_WAIT)
 java      21591               freeacs   76u     IPv6           13053443       0t0        TCP localhost:54167->localhost:https (CLOSE_WAIT)
 java      21591               freeacs   77u     IPv6           13055643       0t0        TCP localhost:59847->localhost:https (CLOSE_WAIT)
 java      21591               freeacs   78u     IPv6           13054691       0t0        TCP localhost:46345->localhost:https (CLOSE_WAIT)
 root@acs:~# lsof | grep "CLOSE_WAIT" | wc -l
 8736

Last night it was showing as over 28,000 (before stopping the service when testing Monit) — this might be why the service sometimes logs that it has run out of open files. However, this count is not monotonic — it has just increased and then decreased.

HTTP is set (by request!) to redirect to HTTPS, and this is causing the monitor service to “leak” sockets. (I realise that such a redirect makes no sense, but routers can and do retransmit a POST request in response to a 80 to 443 redirect!)

(I don’t actually know what this service even does; looking at the source code I can’t find any evidence that it does anything except log its observations, and when we did have a service hang, it did not restart it (or even notice), so it seems not to be able to take any remedial action. I’m setting up Monit instead, which is made more interesting by every service being called “java” and there being no PID files …)

This is freeacs-monitor 2.1.6 in Ubuntu 18.04.1 x64 LTS.

Daniel-Beardsmore, 2019-04-11:
(Ignore the stupid strikethrough — Github have rejected extension-friendly Firefox so I have neither preview nor edit. For some reason it’s taken #…# as a multi-line comment.)

jarlah, 2019-04-11:
I am definitely aware of the fact that the monitor service is badly implemented. Its actually a very simple task. Check endpoint up status, retry and log. Anyone can contribute if i am not doing it myself. But not sure when i can

closed 2021-09-04 by jarlah