freeacs/freeacs #311: how do i set a cpe's connection request username and password with freeacs

techdpnn, 2019-04-12:
hi, fellas~

techdpnn, 2019-04-12:
i have just installed freeacs in unbuntu 18. 
but how can i set a cpe's connection request username and password?
there are some communications between freeacs and my cpe ,i can see that by wireshark .

acs adress is 192.168.0.103 , cpe is 192.168.0.101
[image]

pingwiniasty, 2019-04-12:
Connection Request username and password are configured in cpe by default, and You need to add cpe manually to ACS and provide correct password. But you can also configure tr069 module with `discovery.mode =  true` and cpe will be added autmaticly

techdpnn, 2019-04-12:
@pingwiniasty  thanks for response.
i have configured discovery.mode = true and my cpe was found in freeacs.
However,they can only do inform communication between cpe and freeacs.
I think , if freeacs want to send message to cpe by `connection request url`, there should be a authorize with username and password. and Wireshark message above proved that

techdpnn, 2019-04-12:
the exact parameter name is `InternetGatewayDevice.ManagementServer.ConnectionRequestURL`

pingwiniasty, 2019-04-12:
Ok. In unit configuration, below colum named "Unit value" select "all" from combobox and try to find ConnectionRequestUsername and ConnectionRequestPassword. If it not exist you can this parameter manually in Unit Type and configure its value in Unit Profile or Unit

techdpnn, 2019-04-12:
i have added these two parameter,and set their flag to 'RW', it  didn't work.
then i tried set flag to "X", Wireshark( a network protocol analyze tool ) tips "unauthorized".
in a word , it didn't work.

pingwiniasty, 2019-04-12:
Could you attach `tr069-conversation.log` here?

techdpnn, 2019-04-12:
[tr069-conversation.log](https://github.com/freeacs/freeacs/files/3072246/tr069-conversation.log)

vasqude, 2019-04-26:
I have a similar question to this. How can we make it so that the ACS automatically populates these values from the CPE?
 I can verify that the CPE is sending the information but I still have to manually create the parameter. What do we need to change so that when a new CPE contacts the ACS the ACS populates the Connection Request username, password, and URL? Other parameters are auto-populated as shown below. 

[image]

jarlah, 2019-04-26:
you can set RA flag on the unittype parameters. Then they will be requested read from the cpe by the acs. Other params are system parameters and are handled specially.

vasqude, 2019-04-26:
Jarlah -

Thank you for your response, the issue that I am seeing now is that once I changed the flag from RW to RA on the ConnectionRequest username, password and URL parameters. The conversation between CPE and  ACS doesn't stop. The TCP dumps show that the 204 code is sent and then immediately the conversation begins all over again. Is this an issue or have you seen this before? It does impact provisioning I cannot send new values since the initial conversation has not ended.

vasqude, 2019-04-29:
I found the issue, the InternetGatewayDevice.ManagementServer.PeriodicInformInterval value keeps changing, therefore, causing the conversation to be stuck in a loop. What do we need to change to make this correction? 

[image]

jarlah, 2019-04-29:
@techdpnn endless tries from the cpe to call GetRPCMethods. Not supported currently. But dont fret :) its supported on master. I just need to stabilize master and make a release. Cant give a date though. But its at least one of the next weeks

closed 2021-09-04 by jarlah