freeacs/freeacs #313: Cannot get FreeACS to see my CPEs

knackrack615, 2019-04-12:
Hello, after many attempts and trying anything i could think of i still can't get my CPEs to show in FreeACS

I've set discovery.mode to true and also set authentication to none

I'm running FreeACS in docker and i've applied the settings both by editing the .env file (which didn't seem to do anything) and by manually editing the application.conf inside the docker container.

The only thing i'm getting in the logs is this: https://hastebin.com/pasirijige.coffeescript (nothing in tr069 related logs)

jarlah, 2019-04-12:
Server does not recognize GetRPCMethods.

knackrack615, 2019-04-12:
Alright, thanks for clarifying!

closed 2019-04-12 by knackrack615

jarlah, 2019-04-23:
@knackrack615 i have now implemented support for both receiving and sending GetRPCMethods. But there is currently no logic to trigger this method to the device in the server. But it will properly receive the request and decide to return a list of methods supported.

jarlah, 2019-04-23:
Not released yet.