freeacs/freeacs #314: script can not excute  successful

sifksky, 2019-04-12:
Hi:
  I create a script named "ls -al /  /tmp/kk.txt". But I can see the log from stun server.
```
2019-04-12 18:59:37 [DBI for STUN] ERROR com.github.freeacs.dbi.DBI - An error occurred in DBI.run() (stacktrace printed earlier): Communications link failure

The last packet successfully received from the server was 1 milliseconds ago.  The last packet sent successfully to the server was 1 milliseconds ago.
Loading class `com.mysql.jdbc.Driver'. This is deprecated. The new driver class is `com.mysql.cj.jdbc.Driver'. The driver is automatically registered via the SPI and manual loading of the driver class is generally unnecessary.
2019-04-12 18:59:55 [main] WARN  c.z.hikari.util.DriverDataSource - Registered driver with driverClassName=com.mysql.jdbc.Driver was not found, trying direct instantiation.
2019-04-12 18:59:57 [Thread-9] WARN  o.e.jetty.server.AbstractConnector - Ignoring deprecated socket close linger time
2019-04-12 19:23:00 [Scheduler STUN] WARN  KickSingle - nfs-kk responded with HTTP 401, indicating a unsuccessful kick
2019-04-12 19:30:30 [Scheduler STUN] WARN  KickSingle - nfs responded with HTTP 401, indicating a unsuccessful kick
2019-04-12 19:30:57 [Scheduler STUN] WARN  KickSingle - nfs did not respond, an error has occured: 
java.net.SocketTimeoutException: Read timed out
        at java.net.SocketInputStream.socketRead0(Native Method)
        at java.net.SocketInputStream.socketRead(SocketInputStream.java:116)
        at java.net.SocketInputStream.read(SocketInputStream.java:171)
        at java.net.SocketInputStream.read(SocketInputStream.java:141)
        at org.apache.http.impl.io.AbstractSessionInputBuffer.fillBuffer(AbstractSessionInputBuffer.java:161)
        at org.apache.http.impl.io.SocketInputBuffer.fillBuffer(SocketInputBuffer.java:82)
        at org.apache.http.impl.io.AbstractSessionInputBuffer.readLine(AbstractSessionInputBuffer.java:276)
        at org.apache.http.impl.conn.DefaultHttpResponseParser.parseHead(DefaultHttpResponseParser.java:138)
        at org.apache.http.impl.conn.DefaultHttpResponseParser.parseHead(DefaultHttpResponseParser.java:56)
        at org.apache.http.impl.io.AbstractMessageParser.parse(AbstractMessageParser.java:259)
        at org.apache.http.impl.AbstractHttpClientConnection.receiveResponseHeader(AbstractHttpClientConnection.java:294)
        at org.apache.http.impl.conn.DefaultClientConnection.receiveResponseHeader(DefaultClientConnection.java:257)
        at org.apache.http.impl.conn.ManagedClientConnectionImpl.receiveResponseHeader(ManagedClientConnectionImpl.java:207)
        at org.apache.http.protocol.HttpRequestExecutor.doReceiveResponse(HttpRequestExecutor.java:273)
        at org.apache.http.protocol.HttpRequestExecutor.execute(HttpRequestExecutor.java:125)
        at org.apache.http.impl.client.DefaultRequestDirector.tryExecute(DefaultRequestDirector.java:679)
        at org.apache.http.impl.client.DefaultRequestDirector.execute(DefaultRequestDirector.java:481)
        at org.apache.http.impl.client.AbstractHttpClient.doExecute(AbstractHttpClient.java:835)
        at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:83)
        at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:108)
        at com.github.freeacs.stun.Kick.kickUsingTCP(Kick.java:153)
        at com.github.freeacs.stun.Kick.kickInternal(Kick.java:86)
        at com.github.freeacs.stun.Kick.kick(Kick.java:57)
        at com.github.freeacs.stun.SingleKickThread.processUnitWatchList(SingleKickThread.java:183)
        at com.github.freeacs.stun.SingleKickThread.run(SingleKickThread.java:243)
        at java.lang.Thread.run(Thread.java:748)
2019-04-12 19:32:51 [Scheduler STUN] WARN  KickSingle - nfs responded with HTTP 401, indicating a unsuccessful kick
2019-04-12 19:36:31 [Scheduler STUN] WARN  KickSingle - nfs responded with HTTP 401, indicating a unsuccessful kick
2019-04-12 19:37:41 [Scheduler STUN] WARN  KickSingle - nfs responded with HTTP 401, indicating a unsuccessful kick
2019-04-12 20:09:47 [Scheduler STUN] WARN  KickSingle - nfs responded with HTTP 401, indicating a unsuccessful kick
2019-04-12 20:10:47 [Scheduler STUN] WARN  KickSingle - nfs responded with HTTP 401, indicating a unsuccessful kick
Loading class `com.mysql.jdbc.Driver'. This is deprecated. The new driver class is `com.mysql.cj.jdbc.Driver'. The driver is automatically registered via the SPI and manual loading of the driver class is generally unnecessary.
2019-04-12 20:28:56 [main] WARN  c.z.hikari.util.DriverDataSource - Registered driver with driverClassName=com.mysql.jdbc.Driver was not found, trying direct instantiation.
2019-04-12 20:28:57 [Thread-9] WARN  o.e.jetty.server.AbstractConnector - Ignoring deprecated socket close linger time

```

"responded with HTTP 401, indicating a unsuccessful kick", I think it is my fault. Someting para do es not config, ple help.

[image]

jarlah, 2019-04-12:
Is the ip to the device accessible from the acs server and can you telnet into the 5000 port on the ACS server?

jarlah, 2019-04-12:
And why do you have /tr069 in the connection request url? That is the connection request url. The url to kick the device. It should not be the url to the tr069 server

jarlah, 2019-04-12:
I may be wrong but i thought connection request url was the connection url to the device. Writing this while on the train. So... But i think i remember correctly

sifksky, 2019-04-12:
Yes, I can telnet 10.70.100.17 5000 successful.
The "/tr069" is our cpe's path. 
[image]

jarlah, 2019-04-12:
Ok.

sifksky, 2019-04-12:
But why the kick return 401, I need config the console password of router ?

jarlah, 2019-04-12:
You got a http 401 which means unauthorized. You got one timeout but i think a random timeout would be acceptable. 401 means wrong username or password. Try setting a password and also set password in unit param. Too see if we have a bug with password less kicks

sifksky, 2019-04-13:
Thank you. I know the cause. The acs can not read the username or password. I should set the flags from R OR RW to RA.

closed 2019-04-13 by sifksky