freeacs/freeacs #315: Use docker-compose

sifksky, 2019-04-13:


sifksky, 2019-04-13:
I use docker-compose install freeacs, can not see pic at dashboard, detail.
[image]
```
2019-04-13 09:41:23 [qtp1702135574-373] ERROR spark.http.matching.GeneralError - 
java.lang.NullPointerException: null
        at sun.awt.FontConfiguration.getVersion(FontConfiguration.java:1264)
        at sun.awt.FontConfiguration.readFontConfigFile(FontConfiguration.java:219)
        at sun.awt.FontConfiguration.init(FontConfiguration.java:107)
        at sun.awt.X11FontManager.createFontConfiguration(X11FontManager.java:774)
        at sun.font.SunFontManager$2.run(SunFontManager.java:431)
        at java.security.AccessController.doPrivileged(Native Method)
        at sun.font.SunFontManager.<init>(SunFontManager.java:376)
        at sun.awt.FcFontManager.<init>(FcFontManager.java:35)
        at sun.awt.X11FontManager.<init>(X11FontManager.java:57)
        at sun.reflect.GeneratedConstructorAccessor4.newInstance(Unknown Source)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
        at java.lang.Class.newInstance(Class.java:442)
        at sun.font.FontManagerFactory$1.run(FontManagerFactory.java:83)
        at java.security.AccessController.doPrivileged(Native Method)
        at sun.font.FontManagerFactory.getInstance(FontManagerFactory.java:74)
        at sun.font.SunFontManager.getInstance(SunFontManager.java:250)
        at sun.font.FontDesignMetrics.getMetrics(FontDesignMetrics.java:264)
        at sun.java2d.SunGraphics2D.getFontMetrics(SunGraphics2D.java:864)
        at org.jfree.chart.text.TextUtils.deriveTextBoundsAnchorOffsets(TextUtils.java:368)
        at org.jfree.chart.text.TextUtils.drawAlignedString(TextUtils.java:331)
        at org.jfree.chart.plot.dial.DialTextAnnotation.draw(DialTextAnnotation.java:305)
        at org.jfree.chart.plot.dial.DialPlot.draw(DialPlot.java:558)
        at org.jfree.chart.JFreeChart.draw(JFreeChart.java:1286)
        at org.jfree.chart.JFreeChart.createBufferedImage(JFreeChart.java:1453)
        at org.jfree.chart.JFreeChart.createBufferedImage(JFreeChart.java:1434)
        at org.jfree.chart.ChartUtils.writeChartAsPNG(ChartUtils.java:184)
        at org.jfree.chart.ChartUtils.writeChartAsPNG(ChartUtils.java:138)
        at com.github.freeacs.web.app.page.unit.UnitStatusPage.getReportChartImageBytes(UnitStatusPage.java:724)
        at com.github.freeacs.web.app.page.unit.UnitStatusPage.getOverallStatusSpeedometer(UnitStatusPage.java:644)
        at com.github.freeacs.web.routes.UnitDashboardRoute.lambda$addRoutes$5(UnitDashboardRoute.java:105)
        at spark.RouteImpl$1.handle(RouteImpl.java:72)
        at spark.http.matching.Routes.execute(Routes.java:61)
        at spark.http.matching.MatcherFilter.doFilter(MatcherFilter.java:134)
        at spark.embeddedserver.jetty.JettyHandler.doHandle(JettyHandler.java:50)
        at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:1568)
        at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:144)
        at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:132)
        at org.eclipse.jetty.server.Server.handle(Server.java:503)
        at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:364)
        at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:260)
        at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:305)
        at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:103)
        at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:118)
        at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.runTask(EatWhatYouKill.java:333)
        at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:310)
        at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.tryProduce(EatWhatYouKill.java:168)
        at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:126)
        at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:366)
        at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:765)
        at org.eclipse.jetty.util.thread.QueuedThreadPool$2.run(QueuedThreadPool.java:683)
        at java.lang.Thread.run(Thread.java:748)

```

pingwiniasty, 2019-04-13:
Attach to container and try to install X11 fonts. In debian/ubuntu package name is `xfonts-base` 
I could check it tomorrow

sifksky, 2019-04-14:
The image is alpine. 
Use: apk update
        apk add xfce4
will fix this issue.Thank you very much.

jarlah, 2019-04-14:
This must be solved for all.

We need to fix it in web/Dockerfile

jarlah, 2019-04-21:
apk add xfce4? like the Xfce Desktop Environment? Nope. There need to be another way :)

closed 2021-09-04 by jarlah