freeacs/freeacs #324: Readall after provisioning is not working

vasqude, 2019-04-24:
Hi -

I have been testing different units with the freeacs software and I cannot get the "read all" to work after I have provisioned a device. The provisioning works fine, but then I want the CPE to send the new configuration to the ACS to verify changes through the freeacs GUI and I get a timeout error. It's only after I reboot the CPE that the read all command that I sent works. 
 
I have attached the Packet captures for each change that I made. Please let me know if you need more information. 
[20190424acs-readall.zip](https://github.com/freeacs/freeacs/files/3112969/20190424acs-readall.zip)

pingwiniasty, 2019-04-25:
Are your device listening on port 8890? Can you reach them in web browser?

vasqude, 2019-04-25:
Yes, I can reach them via HTTP (GUI). The CPE takes the configuration values that I change from the ACS but once the conversation is done. I cannot send another change whether it be a read all or a provisioning command. I get a read timeout error.  
[image]
[image]

pingwiniasty, 2019-04-25:
Please show me `ConnectionRequestURL` displayed in cpe page. Maybe you need set **A** flag for this parameter

pingwiniasty, 2019-04-26:
In your `readall` file, cpe not respond on get request from ACS.
Sometimes I had same problems with some CPEs.

vasqude, 2019-04-26:
yes, you are correct. I have another CPE and the ACS server works as expected with it.

closed 2019-04-26 by vasqude