freeacs/freeacs #329: web services: Documentation

dariosky75, 2019-05-02:
Are there APIs/webservices? I've found very old documentation  that probably does not apply anymore.

With the shell not working I would like to try to automate some tasks using the APIs.

Thanks in advance for your support.
Dario

jarlah, 2019-05-02:
First. Tell me exactly why Shell is not working. And how it behaves. I dont seem to remember or find the issue where you mention that Shell was not working. Please link to it if it exists.

If you tell me what you try to do and what is not working i will see what i can do.

jarlah, 2019-05-02:
We have a soap webservice. And it should work. I will get back to you on how to request its wsdl in the browser.

dariosky75, 2019-05-02:
@jarlah 
Hi tks for the quick response.
I reead somewhere that in the 'porting' the shell module  got broken. Did I got it wrong?
Morover to give you some context looks like I can't find the moveunit command.
What I would like to do is to move Unit belonging to a group from one profile to another
Maybe it's just the documentation that is outdated?
Note: the module itself looks good, see below, snipped form bash

diano@freeacsUbuntu:/opt/freeacs-shell/bin$ ./freeacs-shell
Loading class `com.mysql.jdbc.Driver'. This is deprecated. The new driver class is `com.mysql.cj.jdbc.Driver'. The driver is automatically registered via the SPI and manual loading of the driver class is generally unnecessary.
=== FUSION LOGIN ===
You may skip login to become Admin, just press ENTER
Username: />
2019-05-02 15:23:52 WARN  c.z.hikari.util.DriverDataSource - Registered driver with driverClassName=com.mysql.jdbc.Driver was not found, trying direct instantiation.
  ____ _  _ ____ _ ____ _  _    ____ _  _ ____ _    _
  |___ |  | [__  | |  | |\ |    [__  |__| |___ |    |
  |    |__| ___] | |__| | \|    ___] |  | |___ |___ |___ v2.3.41

  Fusion Shell provdes a variety of commands and possibilities for the advanced
  user of Fusion. It is possible to make scripts to automate tedious tasks, and
  the scripts can be made fairly advanced by the use of file input/output.
  Furthermore it is possible to combine several scripts so the user can build
  script libraries for optimal reuse. To learn all about Fusion Shell you need
  to read Fusion Shell User Manual, but 4 commands is a must at this point:

  - Type 'help' to get context-sensitive help.
  - Type 'help generic' to get generic help.
  - Type 'help <name-of-command>' to get detailed command help.
  - Type 'exit' to exit.

  By using help the user should be able to learn most of the commands in Fusion
  Shell, which will enable him to create/update/delete/inspect all parts of
  the Fusion database. The argument in the 'help' command can be shortened as
  long as it's not ambigious.

jarlah, 2019-05-02:
http://localhost:8088/webservice/ws/acs.wsdl

jarlah, 2019-05-02:
have you read the documentation? I will dig it up.

jarlah, 2019-05-02:
```
=== FUSION LOGIN === 
You may skip login to become Admin, just press ENTER
Username: />

2019-05-02 15:44:22 WARN  c.z.hikari.util.DriverDataSource - Registered driver with driverClassName=com.mysql.jdbc.Driver was not found, trying direct instantiation.
  ____ _  _ ____ _ ____ _  _    ____ _  _ ____ _    _    
  |___ |  | [__  | |  | |\ |    [__  |__| |___ |    |
  |    |__| ___] | |__| | \|    ___] |  | |___ |___ |___ v2.3.41

  Fusion Shell provdes a variety of commands and possibilities for the advanced 
  user of Fusion. It is possible to make scripts to automate tedious tasks, and 
  the scripts can be made fairly advanced by the use of file input/output. 
  Furthermore it is possible to combine several scripts so the user can build 
  script libraries for optimal reuse. To learn all about Fusion Shell you need 
  to read Fusion Shell User Manual, but 4 commands is a must at this point:

  - Type 'help' to get context-sensitive help.
  - Type 'help generic' to get generic help.
  - Type 'help <name-of-command>' to get detailed command help.
  - Type 'exit' to exit.

  By using help the user should be able to learn most of the commands in Fusion 
  Shell, which will enable him to create/update/delete/inspect all parts of 
  the Fusion database. The argument in the 'help' command can be shortened as 
  long as it's not ambigious.

/>help
help

Shows the list of available commands.
Type 'help generic' to get generic commands.
Type 'help <command>' to get detailed help on each command.

	listunittypes [<unittype-name-pattern>]
	listunits [<search-value>]
	setunittype <unittype-name> TR-069|N/A (<vendor> <description>)|<unittype-file> 
	delunittype <unittype-name>
	listparams [staging]
	listusers [<username-pattern>]
	setuser <username> <fullname> <secret> <accesslist>|NULL <admin>
	deluser <username>
	listperms [<unittype-name-pattern>]
	setperm <username> <unittype-name> <profile-name>|NULL
	delperm <username> <unittype-name> <profile-name>|NULL
	listcertificates
	setcertificate <name> <certificate>
	delcertificate <name>

/>setunittype Test TR-069 Test Test
setunittype Test TR-069 Test Test
The unittype Test is added.
/>help cc
help cc
   Syntax  cc <context>                                                                                                   

  Comment  Change to another context. The context change is fairly robust, so you can make context changes directly from, 
           say a unit context  to a group context, if only the group specified is within the same unit type.              

Arguments                                                                                                                 
<context>  Specify context to change to.                                                                                  
           Syntax: .. or /(<type>:<contextname>/)*                                                                        
           Use .. to go one step back in the context hiearchy.. Types are:                                                
           	'ut' (Unit Type)                                                                                              
           	'pr' (Profile)                                                                                                
           	'un' (Unit)                                                                                                   
           	'gr' (Group)                                                                                                  
           	'jo' (Job)                                                                                                    
           	'up' (Unit Type Parameter)                                                                                    
           Use a single '/' to specify root. If context name contains whitespace, enclose the entire context with quotes: 
           "/ut:Test UT/".                                                                                                

 Examples                                                                                                                 
           cc ..                                                                                                          
           cc /                                                                                                           
           cc /ut:TestUnittype/pr:TestProfile                                                                             
           cc "/ut:Test Unit Type/"                                                                                       
           cc /gr:MyGroup                                                                                                 

/>cc ut:Test
cc ut:Test
/ut:Test/>
```
the doc is in the service, but I should maybe add `help cc`as a hint in the first menu ...

jarlah, 2019-05-02:
```

/ut:Test/>help generic
help generic

Shows the list of available commands.
Type 'help generic' to get generic commands.
Type 'help <command>' to get detailed help on each command.

	cc <context>
	unit [<search-value>]
	syslog [<search-args>] [<list-args>]
	call <scriptname> 
	setvar <name> <expression>
	delvar <name>
	listvars
	cat <filename>
	delosfile <filename>
	unittypeimport <unittypename>|ALL
	unittypeexport <unittypename>|ALL
	unittypecompletedelete <unittypename>|ALL
	logout
	dbinfo
	userinfo
	echo <text>
	exit
	help [<command>|generic]
	ls [<directory>]
	sleep [<seconds>]
	pausescript
	error <error-message>

/ut:Test/>
````
nope its all there

help generic and help tells you a lot

jarlah, 2019-05-02:
There is a pdf in the docs folder in Shell module. So if you want you can read it. But the Shell module is basically self documenting.

What has changed is that you cannot any longer change data source dynamically. You are connected to the datasource defined in the Shell property file.

jarlah, 2019-05-02:
There may be some good examples on file redirection in the pdf. But some chapters are now irrelevant

dariosky75, 2019-05-02:
I am afraid my confusion comes from reading the pdf you are referring to, hence the "moveunit" command.
that i can't find in the shell. Am I missing something?
Tks
Dario

jarlah, 2019-05-02:
so, the basic premise is that you have to be in the right spot. But we could maybe make a command map for the shell module. A list of all commands and where they can be used.

jarlah, 2019-05-02:
I will make a wiki page. :) soon.

dariosky75, 2019-05-02:
Thanks a lot Jarl. I see. Clear.
I will go on then with some tests. If I end up with something useful that adds avalue, I will post it

closed 2021-09-04 by jarlah