freeacs/freeacs #331: Docker Stun URL and Ports

vanzylwj, 2019-05-06:


jarlah, 2019-05-19:
I have now updated the docker-compose.yml file in commit https://github.com/freeacs/freeacs/commit/8de081b27a1f21fdd564ae0b697ba2f18d3baaf4 ... you can download it or just copy the config into your own file.

vanzylwj, 2019-06-12:
Dear Jara

It exposes the port to the container, but inside the container it is only listening on 127.0.0.1 and not 0.0.0.0. Port 8080 works as it listens on 0.0.0.0.

Output of netstat.
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN      7/java
tcp        0      0 127.0.0.11:40919        0.0.0.0:*               LISTEN      -
tcp        0      0 172.25.0.7:42410        172.25.0.5:3306         ESTABLISHED 7/java
tcp        0      0 172.25.0.7:8080         172.25.0.4:48649        ESTABLISHED 7/java
tcp        0      0 172.25.0.7:8080         172.25.0.4:55067        ESTABLISHED 7/java
tcp        0      0 172.25.0.7:8080         172.25.0.4:49927        ESTABLISHED 7/java
tcp        0      0 172.25.0.7:42408        172.25.0.5:3306         ESTABLISHED 7/java
udp        0      0 127.0.0.1:3478          0.0.0.0:*                           7/java
udp        0      0 127.0.0.1:3479          0.0.0.0:*                           7/java
udp        0      0 127.0.0.11:51385        0.0.0.0:*                           -

Setting the primary.ip = 0.0.0.0 and primary.ip = 172.25.0.7 (container IP) does not work.

Can you please help me with setting this settings to work.

jarlah, 2019-06-13:
it is solvable. Will look at it.

jarlah, 2019-06-15:
I have managed to expose the ports in docker. I am not sure why, but i think the stun ports have previously been exposed on 127.0.0.1, which doesnt work in docker (hope it works normally outside docker), and i had to override the ips to be 0.0.0.0 to get it working. Tested by issuing ```nc -z -v -u localhost 3478``` in the terminal after which i got ```Connection to localhost 3478 port [udp/*] succeeded!```

You will have to download the .env file from the scripts folder, or just look at it and append the new stun config to your local version of the .env file. Then ```docker-compose -p freeacs stop stun``` and ```docker-compose -p freeacs rm stun``` ... and then ```docker-compose -p freeacs up -d``` which will auto create a new stun server.

Docker images have been updated. issue a ```docker-compose -p freeacs pull``` to update images.

jarlah, 2019-06-15:
I need to consider if i should change the default property value to 0.0.0.0 as i mentioned in https://github.com/freeacs/freeacs/commit/a1c7c3f788df2f366f6c42dcbae0e6ca1e651f01#r33952861 as a note to self

jarlah, 2019-06-15:
Have now set stun primary and secondary ips to 0.0.0.0 in https://github.com/freeacs/freeacs/commit/8dde32622d2a850da0d6cfe925df39af310eae38

closed 2021-09-04 by jarlah