freeacs/freeacs #332: KICK Failed

dariosky75, 2019-05-07:
Hi all, 
First of all: I am not sure this is a bug. 
As there might be my misunderstanding or the CPE I am testing is not working properily

I am getting inconsistent and confusing results in ACS initiated connections i.e (KICK)  (Connection Request): most of the time it fails getting this in the sysylog:

_Added unit parameter System.X_FREEACS-COM.IM.Message with value Kick failed at 18:57:50 - require reboot to initiate provisioning :: TCP/HTTP-kick to http://10.40.56.94:58000 (user:acsUser,pass:acsUser) failed, probably due to wrong user/pass since HTTP response code is 401_

My (partial)understanding is: 
When CPE connects at least onece to ACS, ACS also stores the URL to be used to contact the CPE at a later time, the parameter is ManagementServer.ConnectionRequestURL (this is set to A:Always Read at unit type level), to allow for this valure to be saved per CPE in the freeACS Database and be used at later time.

According to TR-069 specs,  the CPE could generate this URL ramdomly (also username?)  so CPEs do not share the same value, ACS gets to know this value for teh CPE only thanks to the initial CPE to ACS exchange

Are there other parameters that need to be set uo? usernamer? passwords?
What am I missing?

Thanks in advance
Dario

From Sysylog:

dariosky75, 2019-05-15:
Hello all, 
No help at all?
The issue is not clear?

Tks
Dario

jarlah, 2019-05-19:
Hi, sorry for late reply. I will look at the issue

jarlah, 2019-05-19:
Before I delve very deep into this, the kick functionality is totally not worth a penny. I really need to simplify all this code jumping just to make a simple get request with authentication to a device connection request url. Its a problem with the whole micro service enterprisey model of the stack. 

Yes, its possible to dynamically make a request to the device, e.g. dynamically make a connection request url, but I fully regret ever adding new functionality on top of this logic. 

What I suggest is to investigate if your CPE returns ConnectionRequestURL, ConnectionRequestUsername and ConnectionRequestPassword. If it does, when you set the params to always read (A), then kick should just work. If the CPE does not return those parameters, or just the ConnectionRequestURL, then you have hit the point where this stack meets it self in the door. Its a bit difficult to say at least.

I think I need to get some help from @pingwiniasty  on this one. Maybe he can help with some more usable advice.

pingwiniasty, 2019-05-21:
ConnectionRequestURL parameter must be set "RA" flag by default. This is main problem for new users with KICK.
ConnectionRequestUsername and ConnectionRequestPassword are flexible. Some users want to use credentials from CPE, then flags must be "RA". But we are providing cretendials to CPE on first connect, then we are using "RW" flag

dariosky75, 2019-05-23:
Thanks a lot. I've always used the first Approach you mentioned. I will now try with the second one.

closed 2021-09-04 by jarlah