freeacs/freeacs #333: manually add CPE to ACS

vasqude, 2019-05-15:
Hi -

I need clarification on what the ACS server looks for to associate a preconfigured unit and unit type to the actual physical unit.

I would like to build the Unit first in the ACS server then power the unit on and have it pull the parameters I have already set for said unit. 

Thank you for your assistance

closed 2019-05-16 by vasqude

jarlah, 2019-05-19:
I hope you figured it out. Its just adding the unit with same username as the username the device is configured with as acs username, add PeriodicInformInterval (InternetGatewayDevice.ManagementServer.PeriodicInformInterval or Device.ManagementServer.PeriodicInformInterval) to unit type in prior to provisioning it, set a secret  (the systemet parameter Secret) for the CPE and then powering it on. It should successfully authenticate if username and password matches unitId and secret value. It should then provision sucessfully and set a semi random new periodic inform interval.

jarlah, 2019-05-19:
Next, you can just set the values of any RW parameters you have added in advance of the provisioning, or set Discover system parameter to 1, to discover unit type parameter automatically.