freeacs/freeacs #334: com.mysql.jdbc.Driver

baronkis, 2019-05-23:
Use docker-compose install freeacs and have a problem with mysql jdbc Driver in each containers with database environment. Centos 7 and new docker install. 

Logs say:
Loading class `com.mysql.jdbc.Driver'. This is deprecated. The new driver class is `com.mysql.cj.jdbc.Driver'. The driver is automatically registered via the SPI and manual loading of the driver class is generally unnecessary.
2019-05-23 20:29:23 [main] WARN  c.z.hikari.util.DriverDataSource - Registered driver with driverClassName=com.mysql.jdbc.Driver was not found, trying direct instantiation.
2019-05-23 20:29:25 [Thread-5] WARN  o.e.jetty.server.AbstractConnector - Ignoring deprecated socket close linger time

baronkis, 2019-05-28:
I see logs on freeacs_db_1
2019-05-28T05:49:48.748885Z 3 [Note] Aborted connection 3 to db: 'acs' user: 'acs' host: '172.110.2.4' (Got an error reading communication packets)
2019-05-28T05:49:48.748966Z 4 [Note] Aborted connection 4 to db: 'acs' user: 'acs' host: '172.110.2.4' (Got an error reading communication packets)
2019-05-28T05:49:48.791865Z 8 [Note] Aborted connection 8 to db: 'acs' user: 'acs' host: '172.110.2.7' (Got an error reading communication packets)
if I try command from host telnet [db_container_ip] 3306 connecting successful 
if try command nslookup from freeacs_syslog_1 successful resolve mysql db IP address. 
I can't find that is wrong

jarlah, 2019-05-29:
First log message can be ignored. As long as you dont get any direct errors in the log everything should work (the ones you have posted is not error). Is web working? Is anything working?

baronkis, 2019-05-30:
Web, database, nginx are working, but other containers not connect to databases. My containers don't have internet, they can only communicate on freeacs stack. 
My tr069 last logs:
Caused by: org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'getDataSource' defined in class path resource [com/github/freeacs/config/DataSourceConfig.class]: Bean instantiation via factory method failed; nested exception is org.springframework.beans.BeanInstantiationException: Failed to instantiate [javax.sql.DataSource]: Factory method 'getDataSource' threw exception; nested exception is com.zaxxer.hikari.pool.HikariPool$PoolInitializationException: Failed to initialize pool: Could not create connection to database server. Attempted reconnect 3 times. Giving up.
Caused by: com.zaxxer.hikari.pool.HikariPool$PoolInitializationException: Failed to initialize pool: Could not create connection to database server. Attempted reconnect 3 times. Giving up.
The last packet sent successfully to the server was 0 milliseconds ago. The driver has not received any packets from the server.
Caused by: java.net.ConnectException: Connection refused (Connection refused)
2019-05-30 13:26:29 [Thread-4] WARN  com.github.freeacs.Main - Shutdown Hook is running !

jarlah, 2019-06-02:
I will fix this. Some inconsistencies on master regarding property file changes.

jarlah, 2019-06-02:
Nope. I cannot see any problems. I retract my comment. Default compose works produces error messages, yes, thats correct. But those are error messages that come when docker-compose starts the db at the same time. Do docker-compose start db (or similar) to first start db, then do docker-compose up -d

closed 2019-06-02 by jarlah