freeacs/freeacs #335: Continual Periodic Interval Change

vasqude, 2019-06-03:
I have an issue with Periodic Inform Interval the ACS will not stop sending new interval values to the CPE. The moment I set the spread to 0 then the continual updates stop and there is no further conversation between the CPE and the ACS. Is there a way to resolve this or is this a known bug?

[image]

jarlah, 2019-06-03:
The acs is continually "spreading" the devices as they connect and provision. If every unit had the same provisiniong interval the acs would receive spikes of connects for ex in case of power failures. By spreading the devices continually the acs is making sure that the units will not connect at the same time.

vasqude, 2019-06-03:
Right, I agree that the ACS should spread the devices but the CPE will communicate with the ACS the moment it boots after a power failure. The issue I am seeing is that the ACS continuously calculates an interval value and pushes that value to the CPE. This is happening every second causing there to be unnecessary traffic and the session between the CPE and ACS never ends. Shouldn't the ACS calculate the interval parameter once and push it to the device once?

jarlah, 2019-06-03:
Check the event log on the server. Maybe the provisioning is not successful

vasqude, 2019-06-03:
I have attached the logs from the tr069.log file. The CPE is accepting the values given by the ACS, but the ACS continues to send updated periodic interval values to the CPE. Am I missing something?

 Thank you for all of your help. 

[20190603_ACS_periodic_interval.log](https://github.com/freeacs/freeacs/files/3249179/20190603_ACS_periodic_interval.log)

jarlah, 2019-06-04:
I see a continuously SPV-SPVr-EM, SPV-SPVr-EM ... so we need more info. The log you provided is the conversation log. We need to check the event.log and tr069.log. In event.log there is a line that will represent the whole dialog, if it succeeded or failed. in tr069.log we can see if tr069 server decided to return EM because of an error etc.

vasqude, 2019-06-04:
Jarlah -

Thank you for pointing me in the right direction. In the tr069-event log I am continuously seeing the following:

2019-06-04 09:07:23 [qtp314345112-27704] INFO  Event - 127.0.0.1           634ms [IN(4)-INr] [EM-GPV] [GPVr-SPV] [SPVr-EM]                          [1 params set to CPE: InternetGatewayDevice.ManagementServer.PeriodicInformInterval=3493]

I am not sure what the [IN(4)-INr] [EM-GPV] [GPVr-SPV] [SPVr-EM]  error codes mean.

In the tr069 log, I am receiving

2019-06-04 08:36:17 [qtp314345112-27588] WARN  c.g.freeacs.tr069.methods.FADecision - GPVres contained error, try once more and ask for all parameters
2019-06-04 08:39:35 [Cache Cleanup] ERROR c.g.f.base.SessionDataCacheCleanup - SessionData for 44657F-ENT-CXNK00608A51 (key:node0aierbbo5n7v51zbx2bhngtx342933) was removed from cache after timeout - indicate a session which did not terminate correctly

Again thank you for all of your help!

jarlah, 2019-06-05:
i took a look in the code. FADecision means that the ACS has received a fault from the device. But the conversation log you initially provided was from around 13:50 aprox ... the eventlog clip above is from 09:07. The WARN from tr069 log is from 08:36. So they do not have a relevance to each other. What could be interesting is to see whats inside the eventlog at the same time where you saw the repeated SetParameterValues around 13:50 - 14:00 etc. In addition it would be interesting to see if there are any errors or warnings in that same time period 13:50-14:00. 

Maybe look in wireshark, to see if the device is not satisfied by the acs? Thats my gut feeling, that the device keeps connecting because it means it has not completed the provisioning. Well, lets see in the logs if you find them, if not, maybe try with some other device, and try to eliminate issues that way :)

closed 2021-09-04 by jarlah