freeacs/freeacs #336: Package com.github.freeacs.ws.xml not found in the webservice module

btapo, 2019-06-07:
@jarlah I have been following this project since 2014 but then lost touch.
I appreciate the changes you have made.
I was trying to build the project but the `webservice` module compilation failed.
The directory `com.github.freeacs.ws.xml` was not present in the `webservice` module.

Can you please verify if I am missing something?

btapo, 2019-06-07:
Closing this as I found the files are generated based on XSD configuration.
Running `mvn package` in the `webservice` module generated the files here `/path-to-free-acs-code/freeacs/webservice/target/classes/com/github/freeacs/ws/xml`

closed 2019-06-07 by btapo

jarlah, 2019-06-07:
Nice to see you figured it out. I am trying everyones patience with all my changes, but good to see someone clings onto the hope that all these changes are for the better. Which i am convinced of myself, but its not always easy to transfer this to others :)