freeacs/freeacs #337: DIR-825

baronkis, 2019-06-07:
I run Ubuntu 16.04.6 LTS with latest freeacs, D-LINK DIR-825 router it does not create a conversation with ACS server. Other vendors it's ok also D-LINK DIR-615 K2 is ok. only this model have a problem. I dump packets on server side and show xml in packets. I attached xml file.
[dir-825.xml.txt](https://github.com/freeacs/freeacs/files/3267134/dir-825.xml.txt)
tail -f  /var/log/nginx/access.log
5.20.223.69 - - [07/Jun/2019:21:02:51 +0300] "POST /tr069 HTTP/1.1" 401 12 "-" "CPE"

jarlah, 2019-06-13:
I will start up a local dev server and post this inform and see what happens. Not sure exactly when i can do it but i will make a try.

baronkis, 2019-06-18:
I started on the docker freeacs server and see the same problem with DIR-825. Other devices discovering well.

jarlah, 2019-06-19:
For the case with discovery.mode = false this will be logged:

```
2019-06-19 22:32:44 [qtp1721889604-45] WARN  com.github.freeacs.tr069.SessionData - Did not find unit in unit-table, nothing exists on this unit
2019-06-19 22:32:44 [qtp1721889604-45] ERROR c.github.freeacs.tr069.Provisioning - Authentication failed because unitid was not found (CPE IP address: 0:0:0:0:0:0:0:1) (username: unknown)
```

jarlah, 2019-06-19:
Actually, do you see the line above where it says

```
2019-06-19 22:22:25 [qtp1963701334-40] WARN  c.g.f.base.http.DigestAuthenticator - Authentication not verified, but accepted since in Discovery Mode
```
The fact that it says `DigestAuthenticator` is just a copy paste error. Its actually the BasicAuthenticator that is speaking ...

baronkis, 2019-06-20:
My cat /var/log/freeacs-tr069/tr069-conversation.log is empty then discovery.mode=true and discovery.mode=false. I changed auth.method to none but it didn't. 
I installed new VM with 2.1.8 branch of freeacs, changed discovery and auth method, but same problem - conversation log is empty
My both CPE are with ACS username and pasword. 
I captured packets server side, DIR-825 and DIR-615 witch is oldest model, but work fine.
First conversation is equal, both CPE send XML to SRV
SRV return HTTP/1.1 401 Unauthorized\r\n 
Second conversation CPE (DIR-615) send XML to SRV and I show  line 
Authorization: Basic YWRtaW46YWRtaTAx\r\n
Credentials: test-username:test-password
SRV return HTTP/1.1 200 OK\r\n 
but DIR-825 CPE send XML to SRV and SRV return HTTP/1.1 401 Unauthorized\r\n
I tryed mode auth.method = digest. How do CPE know about digest password ?

jarlah, 2019-06-20:
"My both CPE are with ACS username and pasword." i hope you are not creating a user in web and expecting that the cpe is authenticated with this user. The username and password the cpe authenticates with is not a user. Each allowed unit, either by discovery or manually configuration, authenticates against a unit and its unit parameters. The username corresponds to the unitId. The password to the Secret unit parameter.

Btw digest password is on server, and can be changed in properties. The cpe doesnt need to know the digest secret. It is given indirectly to the cpe via complex hashing in the digest protocol. 

I suppose then that your unit that is not allowed to continue sending packets is not sending along the session header to the server for request #2? The server will send a Set-Cookie. The cpe will send whatever cookies the server has set back to the server for each subsequent requests. I suppose you can check the headers sent for the working device and the device that doesnt work

baronkis, 2019-06-25:
I am only filling the CPE side of the username, pasword lines also URL for tr069. My CPE Connection Request Settings is empty. On ACS server unit not created. 
I wrote D-LINK support and I got a new software version with line in XML <ProductClass> "DIR_825" instead of "Router" but it didn't. Now i notice wireshark dump and you are right, my CPE does not send #2 packet with Set-Cookie. 
Maybe need it create quirk or server.servlet.session.cookie.http-only=true change to false ?

baronkis, 2019-07-23:
I have tested the ACS server produced by D-LINK and see that Set-Cookie is not used. 
[Replay_from_SRV-TO_CPE.txt](https://github.com/freeacs/freeacs/files/3423799/Replay_from_SRV-TO_CPE.txt) 
I see nginx log with code 401 and i thinking about Set-Cookie parameter. 
[23/Jul/2019:22:38:40 +0300] "POST /tr069 HTTP/1.1" 401 12 "-" "CPE"
I tried replacing server.servlet.session.cookie.http-only = false, but that didn't work. Is there another way to disable session authorization?

baronkis, 2019-10-24:
I started negotiations with D-LINK to resolve request # 2

closed 2021-09-04 by jarlah