freeacs/freeacs #338: Bug: device_id (InternetGatewayDevice.ManagementServer.Username ) not checked for SQL special chars

dariosky75, 2019-06-13:
I guess I've found a bug. Not easy to encounter it.. but insideous.
The username the CPE is using in connecting with the ACS, which is used as an ID from ACS, is not checked for special SQL chars.   
That is : InternetGatewayDevice.ManagementServer.Username 

Via the free Acs shell, by mistake, I was able to  change the username of a CPE and remotely provision it to something with " ' " in it. This would basically make FreeACS web console to crash always since every attempt to read form some table result in SQL execption. 
The only way around this is to fix this I had to manually remove the entry in the Mysql database.

Anyway any CPE that useses a username with illegal chars in it, would crash Free ACS.

jarlah, 2019-06-13:
Hmm it should technically not fail due to sql injection issues because we use mostly prepared statements. But maybe i have missed it somewhere. This is very easy to reproduce and fix. But i need to think about om what level i should fix it. Should i always replace bad characters, use another way of using sql or what? Will look at it but not sure when.

dariosky75, 2019-06-13:
I do not think this is urgent. Just wnated to trace it.
BR,
Dario

jarlah, 2019-06-20:
Not sure if i will ever solve this. Closing

closed 2019-06-20 by jarlah