freeacs/freeacs #339: Duplicate TR-069 conversation logging

Daniel-Beardsmore, 2019-06-17:
freeacs-tr069 logs the same information into both /var/log/freeacs-tr069/tr069-conversation.log and /var/syslog, except that the copy in syslog is a mess.

This means that if any situation results in excess logging, you get duplicate levels of excess logging.

freeacs-tr069 should not be writing conversation log entries into syslog: they are not suitable content for syslog (which does not handle multiline messages) and it just creates clutter.

As a workaround to stem the flow of madness I tried stopping all syslog logging for freeacs-tr069, but none of the following work in /opt/freeacs-tr069/config/application-config.conf:

syslog.server.host =
(illegal)

syslog.server.host = ""
(just read as localhost, so the flooding resumes)

syslog.server.host: ${?SYSLOG_SERVER_HOST_NONE}
(invalid)

Removing both syslog.server.host lines is also illegal.

jarlah, 2019-06-17:
possibly due to the fact that console logging is enabled in logback.xml under the config folder (I think its there). 

in tr069 logback xml:

```
    <root level="INFO">
        <appender-ref ref="ASYNC_FILE"/>
        <appender-ref ref="ASYNC_STDOUT" />
    </root>
```
remove the last appender-ref ... so it only logs to file.

jarlah, 2019-06-17:
I should probably disable console logging in all modules, and have file logging be default. But then you want be able to see what's going on when running the application from the IDE. So maybe find a way to have a prod config that is bundled, and use a dev config for testing.

Daniel-Beardsmore, 2019-06-18:
This should be in <logger name="Conversation" level="INFO" additivity="false">, right? This uses ASYNC_CONVERSATION_FILE and ASYNC_STDOUT, and disabling ASYNC_STDOUT here has no effect. Neither does disabling ASYNC_STDOUT from <logger name="Event" level="INFO" additivity="false">.

I know that one of the configuration files is read from the .jar directly and not from the copy in /opt (from past experience trying to fix problems), and I have a feeling that it was indeed the logging configuration that is not editable. This is freeacs-tr069 2.1.8.

jarlah, 2019-06-18:
And you have of course restarted with systemctl? . Ok. Will check.

Daniel-Beardsmore, 2019-06-18:
With the service command, but yes. I’m coming up blank trying to determine what it is doing; even with strace -f I can’t figure out where it reads the log configuration from. Github search can’t find where -Dlogging.config goes.

jarlah, 2019-06-18:
I will solve it :) if its solveable. Just give me some time.

mstrasser2, 2019-11-18:
is there a solution for this issue?
is it possible to simply stop the logging ?

Daniel-Beardsmore, 2019-11-18:
I am using:

/etc/rsyslog.d/00-block-freeacs-spam.conf:
if $programname == 'freeacs-tr069' then stop
if $programname == 'freeacs-monitor' then stop

This causes rsyslogd to simply ignore any log entries coming from those two services.

mstrasser2, 2019-11-18:
thank you very much.
This worked for me!

closed 2021-09-04 by jarlah