freeacs/freeacs #340: some improvement suggestions

yrong, 2019-07-26:
1. the logic seems that every time cpe sends inform, we will invoke GetParameterValues method to refresh cpe,  and then get matched job to do config operations like SetParameterValues. I think it is too complicated and usually value in cpe does not change that often, not necessary to refresh cpe every time, maybe by default when receiving empty message from cpe we can poll job to do the configuration operations.

2. for download,reset,restart,ipping operation just invoking the specific rpc method is not enough. E.g for download operation we should wait for the cpe sending 7-transfer-complete event to deem the operation as success. Also for ipping we should wait for the 8-diagnostic-complete event to deem success. I think for the framework we should abstract some concepts like cwmp-service. one service could contains multiple atomic rpc methods and could crossing multiple cwmp sessions.

jarlah, 2019-08-26:
Nice to see someone take initiative. However i am a bit overloaded at the time. So if you want to contribute send me an email. I dont have the time to manage Reading and replying on github atm.

jarlah, 2019-10-03:
I am back. I fully agree about #1. I am also agreeing on #2. However. I am in a thinking box atm on future progress.

closed 2021-09-04 by jarlah