freeacs/freeacs #341: How to reboot device?

freedomwarrior, 2019-07-26:
Hello. 
I enabled tr069 discovery mode, and I see my test router.
Simple stuff, how can I reboot this router from freeacs panel?

yrong, 2019-07-29:
unit configuration->reboot
http://localhost:8081/web/index?page=unit-configuration&init_refreshpage=Refresh

freedomwarrior, 2019-07-29:
I try, but there is no effect. Did I need to configure something more than auto_discover?
[image]

yrong, 2019-07-30:
you need to start stun server and be sure that cpe support tr111

freedomwarrior, 2019-07-30:
My device supports only tr069. So without tr111 supports, I can`t do some basics stuff like reboot/support etc?

yrong, 2019-07-30:
then you will wait until cpe sending 2 periodical inform to  do such things

yrong, 2019-07-30:
Btw, the current freeacs does not support send tcp connection request to cpe directly. In my forked version I've add some logic for that and feel free to test.
https://github.com/freeacs/freeacs/pull/342/commits/a8458c2b319c7680afd39998481a659f60edac7a

freedomwarrior, 2019-07-31:
Ok, thanx a lot.

closed 2019-07-31 by freedomwarrior