freeacs/freeacs #345: Issue with BOOTSTRAP handling

ChandraJavalkar, 2019-08-14:
Hi,
I am new to FreeACS and somewhat mid-level in the understanding of TR-069.
I have a CPE that I have created as a unit type as follows.
On getting BOOTSTRAP message from the unit, the server is SPVing the RW parameters.
The CPE, since it is a stock model and has no configurations on it, is responding with an error(9005), following which the ACS sends a GPV for Device. which the CPE is responding with an error(9002). This cycle goes on. And I see this error "Kick failed at 06:17:17 - require reboot to initiate provisioning :: Neither a public ConnectionRequestURL nor any UDPConnectionRequestAddress was found".

The question is, is there a way to disable the GPV of the RW parameters. From the documentation, I understood that FreeACS GPVs the configurable parameters and SPVs only those whose values differ from those in db. How can I tell FreeACS to just configure them all without the prior step of GPV(read)ing them. The CPE vendor's ACS works this way.
Appreciate your help!
Thanks,
-Chandra

log(I am unable to enable DEBUG, but that's a different problem)
-----
2019-08-14 14:37:52 [qtp47739397-955] WARN  c.g.freeacs.tr069.methods.FADecision - GPVres contained error, try once more and ask for all parameters
2019-08-14 14:38:48 [qtp47739397-957] WARN  c.g.freeacs.tr069.methods.FADecision - GPVres contained error, try once more and ask for all parameters

conversation log
---------------------
2019-08-14 14:37:52 [qtp47739397-953] INFO  Conversation - ============== FROM CPE ( 000xxx-LTE_Enterprise_C-RANSC_Cntrl-000xxx518308 ) TO ACS ===============
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://schemas.xmlsoap.org/soap/envelope/" xmlns:SOAP-ENC="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:cwmp="urn:dslforum-org:cwmp-1-0"><SOAP-ENV:Header><cwmp:ID SOAP-ENV:mustUnderstand="1">1</cwmp:ID></SOAP-ENV:Header><SOAP-ENV:Body SOAP-ENV:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><cwmp:Inform><DeviceId><Manufacturer>Commscope</Manufacturer><OUI>0005B9</OUI><ProductClass>LTE_Enterprise_C-RANSC_Cntrl</ProductClass><SerialNumber>000xxx518308</SerialNumber></DeviceId><Event xsi:type="SOAP-ENC:Array" SOAP-ENC:arrayType="cwmp:EventStruct[1]"><EventStruct><EventCode>0 BOOTSTRAP</EventCode><CommandKey></CommandKey></EventStruct></Event><MaxEnvelopes>1</MaxEnvelopes><CurrentTime>2019-08-14T12:37:52</CurrentTime><RetryCount>0</RetryCount><ParameterList xsi:type="SOAP-ENC:Array" SOAP-ENC:arrayType="cwmp:ParameterValueStruct[11]"><ParameterValueStruct><Name>Device.ManagementServer.ParameterKey</Name><Value xsi:type="xsd:string">None</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.ConnectionRequestURL</Name><Value xsi:type="xsd:string">http://10.10.30.79/acscall</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.Manufacturer</Name><Value xsi:type="xsd:string">xxxx</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.ManufacturerOUI</Name><Value xsi:type="xsd:string">000xxx</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.ProductClass</Name><Value xsi:type="xsd:string">LTE_Enterprise_C-RANSC_Cntrl</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.SerialNumber</Name><Value xsi:type="xsd:string">000xxx518308</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.HardwareVersion</Name><Value xsi:type="xsd:string">750742.00.04</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.SoftwareVersion</Name><Value xsi:type="xsd:string">1.5.03.143</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.ProvisioningCode</Name><Value xsi:type="xsd:string"></Value></ParameterValueStruct><ParameterValueStruct><Name>Device.IP.Interface.1.IPv4Enable</Name><Value xsi:type="xsd:boolean">1</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.IP.Interface.1.IPv4Address.1.IPAddress</Name><Value xsi:type="xsd:string">10.10.30.79</Value></ParameterValueStruct></ParameterList></cwmp:Inform></SOAP-ENV:Body></SOAP-ENV:Envelope>

2019-08-14 14:37:52 [qtp47739397-953] INFO  Conversation - =============== FROM ACS TO ( 000xxx-LTE_Enterprise_C-RANSC_Cntrl-000xxx518308 ) ============
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
<soapenv:Header>
        <cwmp:ID soapenv:mustUnderstand="1">1</cwmp:ID>
</soapenv:Header>
<soapenv:Body>
                <cwmp:InformResponse>
                        <MaxEnvelopes>1</MaxEnvelopes>
                </cwmp:InformResponse>
</soapenv:Body>
</soapenv:Envelope>
2019-08-14 14:37:52 [qtp47739397-954] INFO  Conversation - ============== FROM CPE ( 0005B9-LTE_Enterprise_C-RANSC_Cntrl-000xxx518308 ) TO ACS ===============

2019-08-14 14:37:52 [qtp47739397-954] INFO  Conversation - =============== FROM ACS TO ( 000xxx-LTE_Enterprise_C-RANSC_Cntrl-000xxx518308 ) ============
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
<soapenv:Header>
        <cwmp:ID soapenv:mustUnderstand="1">FREEACS-1565786272682</cwmp:ID>
</soapenv:Header>
<soapenv:Body>
                <cwmp:GetParameterValues>
                        <ParameterNames soapenc:arrayType="xsd:string[20]">
                                <string>Device.ManagementServer.ConnectionRequestURL</string>
                                <string>Device.ManagementServer.ConnectionRequestUsername</string>
                                <string>Device.ManagementServer.PeriodicInformInterval</string>
                                <string>Device.Services.FAPService.1.CellConfig.LTE.EPC.PLMNList.1.Alias</string>
                                <string>Device.Services.FAPService.1.CellConfig.LTE.EPC.PLMNList.1.Enable</string>
                                <string>Device.Services.FAPService.1.CellConfig.LTE.EPC.PLMNList.1.IsPrimary</string>
                                <string>Device.Services.FAPService.1.CellConfig.LTE.EPC.PLMNList.1.PLMNID</string>
                                <string>Device.Services.FAPService.1.CellConfig.LTE.EPC.TAC</string>
                                <string>Device.Services.FAPService.1.CellConfig.LTE.RAN.PHY.PRACH.RootSequenceIndex</string>
                                <string>Device.Services.FAPService.1.CellConfig.LTE.RAN.RF.DLBandwidth</string>
                                <string>Device.Services.FAPService.1.CellConfig.LTE.RAN.RF.EARFCNDL</string>
                                <string>Device.Services.FAPService.1.CellConfig.LTE.RAN.RF.EARFCNUL</string>
                                <string>Device.Services.FAPService.1.CellConfig.LTE.RAN.RF.PhyCellID</string>
                                <string>Device.Services.FAPService.1.CellConfig.LTE.RAN.RF.ULBandwidth</string>
                                <string>Device.Services.FAPService.1.FAPControl.LTE.AdminState</string>
                                <string>Device.Services.FAPService.1.FAPControl.LTE.Gateway.S1SigLinkServerList</string>
                                <string>Device.Services.FAPService.1.FAPControl.LTE.Gateway.SecGWServer1</string>
                                <string>Device.Services.FAPService.1.FAPControl.LTE.RFTxStatus</string>
                                <string>Device.Services.FAPService.1.X_xxxx_LTE.TACList</string>
                                <string>Device.Time.LocalTimeZone</string>
                        </ParameterNames>
                </cwmp:GetParameterValues>
</soapenv:Body>
</soapenv:Envelope>
2019-08-14 14:37:52 [qtp47739397-955] INFO  Conversation - ============== FROM CPE ( 000xxx-LTE_Enterprise_C-RANSC_Cntrl-000xxx518308 ) TO ACS ===============
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://schemas.xmlsoap.org/soap/envelope/" xmlns:SOAP-ENC="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:cwmp="urn:dslforum-org:cwmp-1-0"><SOAP-ENV:Header><cwmp: [... cut ...] /soap/encoding/"><SOAP-ENV:Fault><faultcode>SOAP-ENV:Client</faultcode><faultstring>CWMP fault</faultstring><detail><cwmp:Fault><FaultCode>9002</FaultCode><FaultString>Internal Error</FaultString>**<SetParameterValuesFault><ParameterName>Device.</ParameterName><FaultCode>9002</FaultCode><FaultString>Internal Error</FaultString>**</SetParameterValuesFault></cwmp:Fault></detail></SOAP-ENV:Fault></SOAP-ENV:Body></SOAP-ENV:Envelope>

[image]
[image]
[image]
The unit as below,
[image]
[image]
[image]

jarlah, 2019-10-03:
You and me have chatted on email. The problem you are facing is that your device is incompatible. It dosent support GPV in bootstrap mode. I need to think about the whole state of freeacs to give a proper Feedback on how it can be best solved. Give me some time and well see.

ChandraJavalkar, 2019-10-04:
Thanks Jarl. Appreciate it!

closed 2021-09-04 by jarlah