freeacs/freeacs #350: Digest Authentication

mstrasser2, 2019-10-10:
i get following error after i update my FritzBox OS form 7.00 to 7.12

ERROR c.github.freeacs.tr069.Provisioning - Digest challenge response does not contain all necessary parameters (CPE IP address: 194.208.205.14) 
(username: 00040E-444E6DC3FC8B", 
realm="xaps", 
nonce="858b601a6da82753676895eblabla", 
uri="/tr069/", 
qop=auth, 
nc=00000001, 
cnonce="9cb0e325e8ebla", 
response="2896290748b5580e506786cbfc9blabla", 
opaque="16e4c40ea15eacbd6041b973blablabla", algorithm=MD)

but when i google this error i get following java code regarding this problem

if ((username == null) || username.length() < 6 || (realm == null) || (nonce == null) || (uri == null) || (response == null))
         throw new TR069AuthenticationException("Digest challenge response does not contain all necessary parameters (CPE IP address: " + reqRes.getReq().getRemoteHost() + ") (username: "
               + username + ")", null, HttpServletResponse.SC_FORBIDDEN);

but all conditions are met to not get this error.
why does this happen?

Thank you
br,
Marcus

pingwiniasty, 2019-10-16:
Could You attach tcpdump/wireshark pcap with device and acs conversation?

mstrasser2, 2019-10-17:
tshark pcap and the tr069.log part

thank you

[acs.zip](https://github.com/freeacs/freeacs/files/3737586/acs.zip)

mstrasser2, 2019-10-22:
@pingwiniasty 

in cooporation with avm (FritzBox) we got the difference

before update 7.10 the digest header looked like this:

Authorization: Digest username="00040E-xxxxxxxxxxx", realm="xaps", nonce="50f0375f8f912341234", uri="/tr069/", qop=auth, nc=00000001, cnonce="07e8e50312341234", response="a6c84d423639dab82ae40ec123412341234", opaque="2dedd1bd011ec0021234123412341234"

with update 7.10 and higher:

Authorization: Digest username="00040E-xxxxxxxxxx", realm="xaps", nonce="9f0b06e63bc7eb46b9123412341234", uri="/tr069/", qop=auth, nc=00000001, cnonce="8b605b0c66f912341234", response="d82a87a60ba9e62557123412341234", opaque="3d263882868bd8e04435792dd5093ec4", **algorithm=MD5**

it looks like that the part "algorithm=MD5" is responsible for that error and therefor the cpe is not able to communicate with the acs!

Does someone know what i can do or is it possible to adapt that part in the java code?

pingwiniasty, 2019-10-22:
Which version of freeacs are you using? In current master branch and versions >= 2.x authentication is handled by spring libraries.

mstrasser2, 2019-10-22:
on our production site the old freeacs is running with tomcat7
and on the test server we have the freeacs version 2.2.0

both verions have the digest authentication problem

mstrasser2, 2019-11-04:
i resolved the issue with haproxy

mstrasser2, 2019-11-04:
i cut the authorization header with 
http-request replace-header Authorization (.*)', algorithm=MD5' \1

closed 2019-11-04 by mstrasser2