freeacs/freeacs #351: Seeking maintainer(s)!

jarlah, 2019-10-12:
I have thought about whether i have enough time and energy to continue the project. 

Sorry to say i concluded that i havent. 

Seeking one or more developers that can continue this project. As long as the License is kept as is.

jarlah, 2019-11-04:
Maintainers have been found. Lead maintainer is now @pingwiniasty. I will still be in the background helping maintainers do what they need to do. But i will not enforce any decisions.

pingwiniasty, 2019-11-04:
Hi! Thanks @jarlah for entry post :D 

Currently with @indiwidu we want to recreate whole web package. There will be full REST API on the backend and Vuejs SPA on the frontend. 
This could contain breaking changes with current version, but we want to make work with freeacs easiest.

Give us time so we can prepare a list of changes and things to do with new version.

yrong, 2020-04-28:
is this project still alive?  I can provide help if needed

jarlah, 2020-04-28:
@yrong I usually never check this project anymore, so you were lucky now ;) . in theory @pingwiniasty took the challenge. However, all of us have a day job. If you or @pingwiniasty, or anyone else, can make fixes and open pull requests however, its not too time consuming for me the briefly code review it and merge it and possibly make a release from time to time.

closed 2021-09-04 by jarlah