freeacs/freeacs #352: Kick Failed

tolukosimi, 2019-10-16:
Hello,

I have been having some issues with setting up the ACS Server. I am a new user of tr069 and have read up everything i could come about on this FreeACS server on git wiki.

I have my device connected to the ACS server and whenever I click on ReadAll under Provisioning, the page reloads without any issue but I still think that something is wrong because anytime I click on Reboot, I get the error message 

"Kick failed at 10:36:35 - require reboot to initiate provisioning :: TCP/HTTP-kick to http://server-address-url/tr069/ failed because of an unexpected error: No route to host (Host unreachable)"

The parameter under connection request URl is the same parameter under the TR069 settings on the device and the connection url credentials is same too. I have tried using the WAN ip address of the device as the connection url but also came back with errors connection refused.

I there something I am not doing right?

pingwiniasty, 2019-10-16:
Are you able to ping this device from ACS host?

tolukosimi, 2019-10-16:
Yes.

I was able to ping the device from the ACS Server and I was able to ping the ACS server from the Device Web Interface tool

pingwiniasty, 2019-10-16:
ConnectionRequestURL must be an URL provided by Device, normally there is an IP address with port or enpoint. If I understand, you have in this parameter ACS URL?

tolukosimi, 2019-10-16:
the ConnectionRequestURL was inserted manually when it failed to work.

I have tried working around it.

During installation, all I did was to run the script on CentOS and everything installed. Nothing else done except to enter true in discovery property.

I entered the server ip, username and password created on the device and i expect it to connect. on checking, I noticed that it already connected and also brought out the parameters... I tried sending a read all command or reboot through the server but always came back with URL error so I entered the URL manually to sort that out ... that is my setup.

I have read everything online and nothing is giving a straight forward setup advice... My device is connected to the internet and the acs server can connect to it as they can ping each other. 

What am i gettig wrog?

pingwiniasty, 2019-10-16:
Go to Unit Type and set ConnectionRequestURL flags to RA (Read, Always-Read). Then remove overrided parameter from Device and reboot it. 
Are you have running STUN app from freeacs? It's listening default on port 8087. If I remember it must be running to preform kicks

tolukosimi, 2019-10-16:
I did not configure STUN as I only need the tr069 on it. Where can I set those?

pingwiniasty, 2019-10-17:
Please check in process list if you have stun.jar running. If not, please run it and check kick.
It must be running.

tolukosimi, 2019-10-18:
It is running.

Please note that I have reinstalled using Ubuntu Server following the steps listed on the main doc.

Still experiencing the same issue.

tolukosimi, 2019-10-18:
Also, I have entered the stun settings and it kick is still not working.

I tried to upload a new software, it stopped half way through without
uploading successful...

pingwiniasty, 2019-10-18:
Could you provide tcpdump pcap when you trying to kick device?

tolukosimi, 2019-10-23:
I have been able to resolve this by reinstalling the application on Ubuntu and following the process all over again.

closed 2021-09-04 by jarlah