freeacs/freeacs #353: Manipulating Strings in Shell Scripts

tolukosimi, 2019-10-23:
Hello,

I am trying to set some parameters using part of the serial number of the device. I was able to set the parameter using the serial number but whenever i try to use a subset of the string, it does not work.

The variable is set using the following line;

var sn '${_System.X_FREEACS-COM.Device.SerialNumber}'

And I can set using the code

setparam InternetGatewayDevice.ManagementServer.Username STANDARD-${sn}

thereby giving it the username STANDARD-serialnumber

I might be wrong, but I think if i want to use part of the string, I can manipulate as thus;

setparam InternetGatewayDevice.ManagementServer.Username STANDARD-${sn:6}

which is supposed to give STANDARD-number 

But instead it is giving me STANDARD-sn:6

What am I doing wrong?

pingwiniasty, 2019-10-28:
You can use substring

`var newSn '${sn}'.substring(0,6)`

tolukosimi, 2019-10-31:
Thank you. This works

closed 2019-11-06 by pingwiniasty