freeacs/freeacs #354: not issue, but about the function about "read all" button on web unit UI

heedong-pixelro, 2019-10-30:
I want to know the function about "read all" button on web unit UI.
I intuitively thought that web UI will display the latest values about parameters of a CWMP after clicking the button of "read all".
But web UI did not display the latest values about parameters on the CWMP.
I had checked out the values in database(acs) mysql and figured out that the values was not updated
though FreeACS received the latest values in inform message sent from the CWMP.

I checked the below url that your team shared.
https://github.com/freeacs/readme/blob/master/Fusion%20System%20Parameter%20Reference.pdf

Above document says "read all" button is for DEBUG.
Please let me know how to use for DEBUG through using "read all" button.

In detail, when clicking "read all" button, the CWMP processed below steps.
1) A CWMP based on EasyCwmp receives the kick of it from stun.
    : Does the CWMP should reply to the stun message sent from FreeACS's stun server as stun response additionally?
2) The CWMP sends inform message to FreeACS as the event of "6 CONNECTION-REQUEST".
    : 6 CONNECTION-REQUEST event, is it the event for inform message triggered from "read all" button? Do I have to use other event.
3) Above inform message includes the latest values of parameters on the CWMP.
4). FreeACS does not update the latest values of parameters of the CWMP to database.
    : is it correct?

heedong-pixelro, 2019-11-04:
Anyone here to help?

pingwiniasty, 2019-11-04:
I will check it tomorrow on our production.

pingwiniasty, 2019-11-04:
Ok, I've checked source code. 
1. I must check it with device.
2. Yes, you are initiate connection from ACS

Read All is only for debug. Values are stored in cache. Provisioning must be in "REGULAR" mode to write changes to database. Devices doing Inform requests periodically with "2 PERIODIC" event code to ACS. Then ACS responses with GetParameterValues Request to read all parameters.

closed 2021-09-04 by jarlah