freeacs/freeacs #360: Not all parameters displayed after new CPE is discovered with the same ProductClass value but different set of parameters

nikty, 2020-01-13:
Hello.

I have 2 different router models from the same vendor: ROUTER A and ROUTER B.
They both have the same ProductClass "Router", but they differ in supported parameters.
For example, ROUTER A has fewer instances in "WLANConfiguration" then ROUTER B.

So, when the router with more instances in "WLANConfiguration"  (ROUTER B) is discovered after the ROUTER A, FreeACS doesn't add those new instances and parameters in "WLANConfiguration", though I see in conversation log all the new parameters.

I tried "unitdiscovery" quirks, all to no avail.

How to make FreeACS display all parameters for the new device?

pingwiniasty, 2020-01-14:
You need to create separate unit types for this devices.  Or maybe your devices has different HW version, then you can enable `unit.type.append-hw-version = true` in tr069 config. Then devices will be groupped by name with hw version https://github.com/freeacs/freeacs/issues/12

nikty, 2020-01-14:
I need discovery mode, so creating unit types is not going to help me, right?

I'll try #12 as soon as I'll be ready to build FreeACS from source.

Wouldn't it be more useful to add option to append "InternetGatewayDevice.DeviceInfo.ModelName" and even "InternetGatewayDevice.DeviceInfo.Manufacturer"?

closed 2021-09-04 by jarlah