freeacs/freeacs #372: issue with shell command "provision"

cuongdoannguyen, 2020-11-10:
I have Freeacs 2.2.0 running on Centos8.0 server and mysql cluster. According my view Freeacs 2.2.0 works fine. I have tested it using freeacs-web interface. Thank you for Freeacs!

Now I am testing freeacs-shell commands. At some point I encounter an issue with shell command "provision".
the issue is after a "readall" command is executed,  I then alter few unit parameters and execute "provision" command.  
I watch the status and see the message "SUCCESS: Device has connected to Fusion at". It means the provision has done successfully. But the unit isn't provisioned with new parameters changed. 

to solve this I have to wait about 15 minutes until the temporaty "current unit parameters" has been cached out.
It seems look like as long as the "current unit parameters"exists in cache, unit can not be provisioned with new parameters.

hmh, 2020-11-10:
I suggest you retitle this issue to "shell command 'provision' should flush unit parameters cache", then, since you have already found the root cause of the issue, and the workaround... :-)

cuongdoannguyen, 2020-11-11:
Just retitle!
addition: It seems look like provision using web gui does work correctly. web proces does flush the unit parameter cache.

cuongdoannguyen, 2020-11-17:
webservice "KickUnitRequest" also should flush unit parameters cache!

closed 2021-09-04 by jarlah